PrusaMK2Users/3030_Haribo_Edition #8: Varying M6 Screw Lengths for Printed Parts

ar-o-be, 2017-02-27:
# What
Some printed parts require M6x14, some require M6x12 screws. It's not defined anywhere which parts require which screws.

# Proposed Solution
Either one of these two solutions would suffice:

1. The BOM and Assembly Instructions should carefully note the different lengths required for each part.

*OR*

2. The STL files should be updated to be slightly "beefier" so that we can standardize on the M6x14 screws, and eliminate the need to purchase M6x12.

**Known Parts Which Require M6x12**
- [1x_rambo_mount.stl](https://github.com/PrusaMK2Users/3030_Haribo_Edition/blob/master/STL/1x_rambo_mount.stl) **REQUIRES 2**
- [1x_y_motor_mount](https://github.com/PrusaMK2Users/3030_Haribo_Edition/blob/master/STL/1x_y_motor_mount.stl)

Tigwin, 2017-03-01:
I vote for beefier whenever possible. It adds strength in addition to less checking of screw lengths during the install.

jtktam, 2017-03-01:
some parts there is no clearance for beefy parts and we need to use 12mm

check the z rod holders too

Tigwin, 2017-03-03:
z rod holders were fine for me. just the rambo and y motor mount so far.

jtktam, 2017-03-06:
@ar-o-be and i will be working on updating the hardware count and maybe add in location of each bolt as we get the documentation going

jrezin, 2017-07-19:
all the printer can be currently build without ANY m6x14. besides that, this specific size cannot easily found. it is probably better just get ride of it and use m6x12 and m6x16 (easily found all around the world) instead of using m6x14 as standard.