PrusaMK2Users/3030_Haribo_Edition #9: Need Proper Assembly Instructions

ar-o-be, 2017-02-27:
We need explicit step-by-step instructions for assembly from scratch (using an upgrade kit from Prusa) *AND* assembly using an existing MK2 as a donor.

stevendpclark, 2017-02-27:
When my MK2 upgrade kit, ordered last week so I have a while to go, arrives I can help out with this effort, either by starting, updating and/or reviewing the document

jtktam, 2017-03-06:
@stevendpclark it would be great if you can help out

stevendpclark, 2017-03-15:
I've started documenting my build progress within the wiki of my github clone: 
https://github.com/stevendpclark/3030_Haribo_Edition/wiki/AssemblyInstructions 

Work in progress, and I'm more then willing to let it go if ar-o-be is planning better pictures or would like something else.

stevendpclark, 2017-04-18:
Sorry about the delay, I've had various things come up the last month sadly. 

I'm finally ready to get back to documenting my build but was wondering if this is still a desirable within the project considering the news about the upgrade kits?

JohnOCFII, 2017-04-18:
Welcome back!  I think more clear, simple documentation is still needed.  You and @ar-o-be might be able to divide and conquer!

zsocietyGithub, 2019-01-14:
i plan to Scratch-Build this haribo version,  i have issu about dimenson of the frame profile. can i use this manual instruction to assambly the profile from rebeliX [http://reprap4u.cz/navod-na-stavbu-3d-tiskarny-rebelix/](url) this frame using same profile 3030 and use 3d print part from haribo?.

C4, 2019-01-23:
I've used this guide from Zaribo to assemble my Haribo 3030 build. Instructions are pretty good. Reminds me of Prusa / iFixt style instructions. http://zaribo.org/blog/zaribo-manual-assembly-of-the-frame/