PrusaMK2Users/3030_Haribo_Edition #23: Block spacing for Tnuts on L and T brace possibly needs widening.

davvyk, 2017-03-30:
The current spacing between blocks on the T and L braces only allows space for the 14mm Tnuts purchased from Misumi. 

As these are unavailable to many people outside the states and the 14mm size seems to be unique to Misumi it has been suggested a wider spacing of 16mm be used as this would let people use the commonly available 15.8mm Tnuts.

I knocked up STL files to print myself if you want them, they are linked below.

Thanks. 

https://www.dropbox.com/s/ams92czv7z1n96p/16mm%20L%20Brace.stl?dl=0

https://www.dropbox.com/s/fff85nx6iuj757r/16mm%20T%20Brace.stl?dl=0

ar-o-be, 2017-03-30:
@davvyk Would you mind submitting a PR with the STL files and adding a description to the README about the part and why it's useful? Let me know if you need help with PR and I'd be glad to help out!

davvyk, 2017-03-30:
Hi Arobe.

Yeah if you could give me a pointer on doing a pull request. Never really done a ton of work in git before so dont want to go causing any issues.

Thanks

JohnOCFII, 2018-02-10:
Does this issue still remain, or can this Github issue be closed?

dino340, 2018-05-19:
Opened PR to add davvyk's models to the Optional folder, need to verify if this issue extends to other models.

JohnOCFII, 2018-05-22:
Issue resolved by @dino340 PR.  If this needs to be extended, please re-open.

closed 2018-05-22 by JohnOCFII