PrusaMK2Users/3030_Haribo_Edition #30: Precision Linear Shaft and Bearing Questions

doubleoerik, 2017-04-17:
First question/comment is regarding the Misumi shaft. The part number you specify of PSFG8 is a 8mm Outside Diameter (OD) 1045 carbon steel hard chrome plated shaft with a diametrical tolerance of "f8" which means the diameter will nominally be -0.013mm to -0.035mm less than 8mm. It might be worth noting for the folks using the Misumi LMU8 bearings that the Inside Diameter (ID) tolerance on the bearing is 0mm to -0.009mm and the Misumi technical documentation states that they recommend a shaft with a "g6" tolerance of -0.005mm to -0.014mm (https://us.misumi-ec.com/pdf/fa/2012/p1_0265.pdf). I suspect the shaft you selected is appropriate for the IGUS DRYLIN bushings you call out in your BOM but it seems like folks using LMU8 bearings should consider using the equivalent "g6" tolerance shaft part number PSFJ8 (52100 bearing steel shaft that is induction hardened and is also hard chrome plated). Does that make sense? I'm definitely not an expert so please set me straight if I'm missing something.

The second question/comment is regarding the bearings. The baseline Prusa i3 mk2 design essentially uses back-to-back bearings on the X and Z axis (2 on X as part of the extruder assembly and 2 on each Z carriage). Is there any reason not to go with double bearings (LMUW8) in these three locations instead of sticking with the back-to-back configuration? Perhaps you didn't do this because the IGUS DRYLIN bushings don't have a double length option? We're 3D printing upgraded parts anyway so it seems like a true double bearing would be a better option for these locations. Again, I'm no expert so please educate me if I'm missing something.

Thanks and awesome work on this stuff!

JohnOCFII, 2017-04-17:
@doubleoerik - Interesting note on the LMU8 and the shaft diameter.  I ordered the LMU8 bearings, but assumed similar 8mm tolerances to the default LM8UU bearings from Prusa that shipped with the MK2.

The primary developer of the Haribo 3030 ( @jtktam ) suggested that I try the double-length Misumi bearings in Z - I don't recall if he was also OK with using them in the X carriage.  But - as you point out, I wonder if there will be a tolerance issue.  

You may get more discussion going in the Slack group.  If you aren't already a member, I can send you an invitation.

John

doubleoerik, 2017-04-17:
An invitation would be great - thanks!

JohnOCFII, 2017-04-17:
Sure thing, @doubleoerik - I need your email address -- I don't see it in your GitHub profile.

doubleoerik, 2017-04-18:
doubleoerik@gmail.com

closed 2017-06-16 by JohnOCFII

doubleoerik, 2017-09-12:
Quick follow-up on this - I went ahead and replaced the back-to-back bearings on my z-carriage with double length Misumi LMUW8 bearings. I had to design new z-carriage brackets as I wanted a precision fit for the 45mm long Misumi bearings and I also wanted to incorporate anti-backlash spring loaded nuts into the bracket. It then made sense to redesign the rest of the z-axis brackets (top and bottom) so everything played together nicely but that's for another thread. I've been printing with this setup for months and couldn't be happier with it so anyone out there hesitant to do this upgrade - it's perfextly fine.

I also replaced the back-to-backs on the extruder with a single double bearing. I had to design a adapter to keep the bearing centered in the OEM x-carriage body but otherwise this is working great as well. Next on my list is to mess with the bed as it's not proportionally as rigid/sturdy as the rest of he machine has become. The first thing I may try to do is to replace the single left side bearing with a double length bearing. I suspect Joseph designed it this way so it isn't over-constrained but I'm hopeful I can get it lined up well and bank on the compliance of the system to absorb any overconstraint. The goal is to eliminate the rocking that can occur on the side of the bed with the single bearing.

JohnOCFII, 2017-09-13:
Great info, @doubleoerik - you might also want to post this in the Haribo3030 channel on Mattermost.  (the community moved from Slack a few months ago.  https://3dprinting.community/login