PrusaMK2Users/3030_Haribo_Edition #32: CodeHemi.com Down?

nrepian, 2017-04-19:
I'd really like to mod my MK2 with these excellent 3030 Haribo extrusions, however I can't seem to be able to join the CodeHemi.com slack channel - the host appears to have been down for a few days. I've tried several times with no luck. Is there another way to access it?

jtktam, 2017-04-19:
we are doing invite only now.. if you send me your email i will send you an invite

nrepian, 2017-04-19:
Thanks, that would be great. I can't see a way to send you a private message on here, is there I way I can send my e-mail address without it appearing in public?

rdsmith24, 2017-04-24:
Can you please send me an invite to the Slack group? @jtktam

JohnOCFII, 2017-04-24:
> Can you please send me an invite to the Slack group? @jtktam

If you leave us your email address, we can send you an invitation, @rdsmith24

rdsmith24, 2017-04-24:
Sent you an email....

bemojo, 2017-04-24:
Just put an order from misumi and I would also like to be invited to slack. Not so familiar with git, how do I send you a private msg with my mail?

JohnOCFII, 2017-04-24:
> Just put an order from misumi and I would also like to be invited to slack. Not so familiar with git, how do I send you a private msg with my mail?

@bemojo - my email address is in my GitHub profile.  Click on my photo, and below that is my email address.

JohnOCFII, 2017-04-25:
Today we announced our discussion forum transition from Slack to MatterMost.  The signup page is open to all (again).

It is at https://3dprinting.community/login

Hope to see you there!

John

closed 2017-05-31 by JohnOCFII