PrusaMK2Users/3030_Haribo_Edition #36: HPI Blitz / Firestorm Titanium Idler Gear Shaft 5x16m #104138 availability

mihec87, 2017-05-21:
Trying to source HPI Blitz / Firestorm Titanium Idler Gear Shaft 5x16m #104138( HPI part number #104138) shows that this part is discontinued and is hard to get by...After some digging around i came to part witch is the same dimension(5x16mm but it is not from titan. .#86947 - IDLER GEAR SHAFT 5x16mm
https://www.hpiracing.com/en/part/86947 if we search for hpi 86947 on ebay/google we get much more hits.. I think it is equal replacement...

Tigwin, 2017-05-21:
I can tell you right now that printing this part out of PETG / PLA works great. I had my doubts, but its working fine.

hungtru, 2017-10-30:
can you point me to where you found the stl for it?

dino340, 2018-01-27:
You can easily make one in any modeling software including tinkercad by making a cylinder with a base diameter of 5mm and a height of 16mm.

JohnOCFII, 2018-02-10:
Updated parts also now on BOM.

closed 2018-02-10 by JohnOCFII