PrusaMK2Users/3030_Haribo_Edition #51: Error in X Motor Mounts

Richardgriff, 2017-07-28:
I have been having some problems with printing the X motor mount and everything I tried resulted in the part breaking.

On looking at the file in mesh mixer I found that there appears to be an error in the model that appears in exactly the same location as the part failures.

[image]

[image]

Please can someone tell me how to fix this or fix the model for me.

Thanks

Richard

jtktam, 2017-07-28:
i will have a look at the model.  the part prints fine for most of the folks who been using it.  what slicer you using?

jtktam, 2017-07-28:
[Z motor mount v2.zip](https://github.com/PrusaMK2Users/3030_Haribo_Edition/files/1183463/Z.motor.mount.v2.zip)

can you try this one and see if it solves your issue

the 0.1 offset shouldn't make the part weak, so i suggest you look at your print settings to ensure it's not causing the break

Richardgriff, 2017-07-28:
Hi tam,

Thanks for the quick reply. I am using cubicon creator which I think is based on a version of cura.. I will have a go printing the file you have send over and will let you know.

I have been browsing though the commits and found that the error seems to have crept in when the model was widened for more clearance and the model rotated.

Would printing the original model cause any issues

Richard

Richardgriff, 2017-07-28:
Using a Cubicon Style

Print settings are as follows
ABS @ 260 degrees
Bed temp 110
0.2mm Layer Height
0.8mm Wall thickness
4 Top and Bottom Layers
30% infill
Cooling Fan OFF

jtktam, 2017-07-28:
let me know if the new model fixed your issue, i will commit it to the branch once it's confirmed

the slight shift of 0.1 shouldn't cause the issue you are seeing.. i think the transition from the large base to the vertical member might be your weak spot.. 

i have no experience with abs, i printed most of the parts with petg

Richardgriff, 2017-07-31:
Hay tam,

I tried printing the v2 version and again encountered the problem but slightly less. I passed the stl though several different slicers
The slicers I used where

Cubicon creator
Sli3r prusa
Sli3r normal
Cura 2.6

Viewing the gcode I found that the cubicon creator is the only on to pick up on the flaw and does not produce the layers as you would expect. All other slicers produce a simalr gcode.

The problem I have is that I need to use cubicon creator as it add custom gcode and setting and crc checks them to prove they have been produced using the cubicon creator tool. This is one of the main reason I was trying to switch to an open source printer.

So I ask kindly please can you upload a version with this fix. I know that for 99.9% of the time nobody would see this issue but I would really like to get my printer up and running and these are the final parts

Thanks

Richard

JohnOCFII, 2018-02-10:
Does this issue still remain, or can this Github issue be closed?

w0dash, 2021-01-22:
This issue remains until today (see the blue overhang):

[image]