PrusaMK2Users/3030_Haribo_Edition #75: source files

tcurdt, 2018-01-30:
It seems like this repo only has the STL files.
Where are the source files?
What am I missing?

Tigwin, 2018-01-30:
Welcome to 3d printing. You'll find that most projects release stl files, Prusa included. Anything based on a stl, would obviously also be a stl.

tcurdt, 2018-01-30:
@Tigwin Indeed everything gets exported to STL in the 3d printing world - but hardly anyone uses that as the source format. 

Prusa uses OpenSCAD to model the parts and then exports them to STLs. [The scad files are also available](https://github.com/prusa3d/Original-Prusa-i3/tree/MK2/Printed-Parts). Other projects use Fusion360 or Blender to model their parts.

SkyDriver2k4, 2018-02-10:
the most parts are on jtktams tinkercad. my own files are on fusion. so if you wish to change anything or get access to it. just ask for it on mattermost ~haribo3030 channel.

closed 2018-02-10 by JohnOCFII