PrusaMK2Users/3030_Haribo_Edition #81: Please provide links to public Tinkercad projects for the .stl files.

metacollin, 2018-04-15:
Here, maybe this will make it a little clearer: If I have a link to a public Tinkercad project, I can click 'copy and tinker', and I get all the original individual objects used to create the model, grouped or ungrouped, and can easily edit them. https://metacollin.com/screens/upshot_ysrrJqiv.png

On the other hand, I lose all of this information and cannot easily make any changes if I instead simply import a raw .stl file into Tinkercad: https://metacollin.com/screens/upshot_kEdUsQXH.png

I wish to fix an issue with the poly count of the L and T brackets, as well as make other improvements without having to manually recreate the model (which in some cases isn't even possible to do), by hand.  

There is no issue preventing you from doing this.  Refusing to is both against the spirit of open source, and in violation of the GPL which this project is subject to.  Please include links to the original tinkercad projects, which have been made public (goto settings -> change private to public).  This allows people to 'copy and tinker' within tinkercad.  Whether or not there exist actual cad files is irrelevant to the compliance of the GPL and there is nothing preventing you from doing this.  Tinkercad has the ability to share projects between users, so please share the Haribo 3030 projects.

jtktam, 2018-04-15:
1. this project has nothing to do with prusa research. none of their files were used and if they were used we asked you to print them from THEIR source files
2. the input files this was based from us from the mk2-x project which has stl as the “source files”
so this continues the trend of that project
3. i released the stl, the complete bom and instructions, which is way more than required by any license..  and i have been a proponent of OSHE.. so don’t accuse me of allege infringements
4. in the future you will find that you get more bees win honey than vinegar.  if you contacted me i would have shared it personally above and beyond what is required .. and others have contributed this way.

closed 2018-04-15 by jtktam

jtktam, 2018-04-15:
and i am not obliged by any license to release files in the format that you feel is needed or on any specific platform/package you feel.. what if i drew the first draft with pencil and paper.. you expect that too? 

btw yes i did do it on paper